philrosenberg/instrument-log #52: Setup for TEAMx

philrosenberg, 2025-03-31:
Issue describing the TEAMx setup in Summer 2025

lecjlg, 2025-04-09:
Wifi controllable c form leads. 
16A CEE weatherproof plug-in WiFi switch/timer 

from [https://toughleads.co.uk/?country=GB](https://toughleads.co.uk/?country=GB)

philrosenberg, 2025-04-30:
scaffold towers ordered from Lewis access https://www.scaffold-tower.co.uk/

philrosenberg, 2025-05-29:
scrubber replaced in licor 29 May 2025

philrosenberg, 2025-06-07:
The flux tower mast was erected. The data didn't initially begin logging and it turned out the logging software needed a manual start. See #55 

We had to add a windsock and some white tape to the tower to make it visible as the field is occasionally used by helicopters. This was placed low enough that it couldn't blow up and hit the sonic or licor.

philrosenberg, 2025-06-07:
Note the hatpro still needs calibrating. Linde delivered a pressurised Dewar of LN2 instead of a tanker to fill up our Dewar's. We needed a hose to use this, which is on order.

philrosenberg, 2025-06-08:
To do list for remaining jobs:

Link WiFi power switch for tower
Test power usage when tower is switched off
Sort protocol for power down if battery power is low
Align cimel
Calibrate hatpro
Set scan pattern for gallion
Investigate gallion problem
Set scan pattern for halo
Set scan pattern and retrieval for hatpro
Compare pressure sensors at Sterzing and Plose
Build frame for solar panels
Deploy energy balance station and soil sensors
Measure alignment of lidars and flux tower sonic
Thank yous for farmers

philrosenberg, 2025-06-13:
Dewar hose arrived and we attempted to calibrate hatpro. But too much had boiled off so it was not possible.

philrosenberg, 2025-06-14:
Yesterday's jobs ticked off were aligning the cimel and building frames for solar panels.

We moved the panels as close to the edge of the fenced area as possible away from the tower. I would have liked them further away, but it was just logistically not possible.

The box containing the ecoflow battery was painted white to help keep it cool.

philrosenberg, 2025-06-14:
The it system at the tower with the science kit turned off uses 20 W

philrosenberg, 2025-06-14:
Below is a copy of an email from Adrian Lock from 4th April 2025 regarding soil sensors. I'm putting it here so we know why we chose these levels and for easy access when I'm in the field.

Jules has soil level thicknesses of 0.1, 0.25, 0.65, 2.0m for its standard 4 soil levels, which would put soil level boundaries at 0.1, 0.35, 1, 3m below the surface.  These soil levels are deemed to be layer averages so the ideal observation depths would be at the mid-point of these levels, so at depths of 0.05, 0.225, 0.675 m below the surface.

philrosenberg, 2025-06-15:
Pyranometers and pyrgeometers installed

philrosenberg, 2025-06-16:
Halo lidar is pointing at 282 Deg magnetic
Hatpro is pointing at 98 Deg magnetic
Halo is pointing at 66 Deg magnetic

philrosenberg, 2025-06-16:
Sonic on flux tower is pointing at 51 Deg magnetic
Sonic on hatpro is pointing at 0 mag.

philrosenberg, 2025-06-17:
There seems to be a data logging issue with the HATPRO #58

philrosenberg, 2025-06-23:
Hatpro LN2 calibration performed on Tues 17th June. It was a little procarious putting the calibration target full of LN2 up on the instrument on the platform, so to reduce risk the target was left in place until all the LN2 was gone. It was removed 21st June. Also Charles changed the retrieval and scan pattern. It was all set up from this date.

philrosenberg, 2025-06-23:
Pressure sensor deployed at Bolzano. The gps sensor was placed on the prssure sensor at 9:54 uk time (8:54 UTC).

philrosenberg, 2025-06-23:
The logger software was updated on the Sterzing pressure logger today at around 9:00 UTC 23/6/25

philrosenberg, 2025-06-29:
Gallion lidar is overheating #60 

I am about to change the Halo Lidar to do a wind scan every hour to compensate

philrosenberg, 2025-06-30:
Data logging stopped on the hatpro yesterday evening. I had tried to stop and restart the software with a longer integration time (but couldn't).